LaserWeb/deprecated-LaserWeb3 #100: Time to process

yuusuf-sallahuddin, 2016-08-27:
Is it possible to add a final console log entry at the end of the job stating "Start Time: 15:30:25" & "End Time: 15:35:17" & "Total Time: 4 minutes 52 seconds"?

krolco, 2016-09-01:
I took a look at this tonight.  It turned out to be tricky, so I don't have a solution yet.
The problem is this: there isn't any clean spot anywhere where there is a known end to a job. I can print starts and stops when gcode is submitted, but it gets enqueued and spooled out over USB or ethernet by the node serial connection, and it's not clear when it's done. I'm still thinking about how to identify when a particular job has finished.

funinthefalls, 2016-09-01:
If you use a G28 at the end of the code, could you have laserweb look to see if the X,Y position is at 0,0 and use that as job end?

openhardwarecoza, 2016-09-01:
Thats a very ineligent solution...

Rather; https://github.com/openhardwarecoza/LaserWeb3/blob/master/public/js/socketcomms.js#L77  -  if queue = 0  then we have flushed the whole queue.

krolco, 2016-09-01:
@openhardwarecoza I looked at that, and it works for reporting when the queue is empty, but it doesn't tell us when an entire job is done. We would also be reporting the finish time of jog commands and home commands with that. I'm not even sure that would work, because we can't quite match up the first command to the draining of the queue -- for instance, if we queue several command in a row, those all look like starts, but only a single end event occurs. I guess it might be possible to add a state variable when a command is enqueued in an empty queue, and then toggle it when the queue drains -- that might solve the start/stop problem, but it still means we would generate those on every move, home, jog, and so on.

openhardwarecoza, 2016-09-01:
Var playing which is set to false in the linked function is already the
flag (:

krolco, 2016-09-01:
could probably work with that then :) I'll give it a try tonight.

krolco, 2016-09-02:
I added #105 to implement this.

openhardwarecoza, 2016-09-02:
Merged....

@yuusuf-sallahuddin you can proceed and git pull, then test

closed 2016-09-02 by openhardwarecoza

yuusuf-sallahuddin, 2016-09-04:
Thanks to all that were involved in adding this feature :)

I've done a test & it works well. My only gripe is that all the OKs sent during the job won't let me scroll back up to see when exactly the job was started. Is it possible to output Start Time & End Time @ the end along with the Processed Time.

See my screenshot: 
[image]

krolco, 2016-09-04:
Hi! No problem; I'll add that, probably tonight. It's a tiny change.
One thing to note (which you may have noticed) is that the "job finished" time may be emitted slightly before the laser actually stops -- the controller board may buffer up a couple of commands, so the command queue may drain before the controller has actually executed the last command. It should be close enough though :)

funinthefalls, 2016-09-04:
Would it be possible to track the elapsed time on an ongoing basis as a cumulative total? I thought that we could keep a running total of the approximate time the laser tube has been used. Would help people plan for maintenace and tube replacements.

krolco, 2016-09-05:
@funinthefalls I can take a look at that also. It will be an approximate time, since I don't plan to differentiate between G0 and G1/2/3 moves -- just the total job time.

openhardwarecoza, 2016-09-05:
Yuusuf asked about job time estimate before. We had https://github.com/openhardwarecoza/LaserWeb3/blob/master/public/lib/gcode-viewer/gcode-parser.js#L1306-L1308 but i recall it broke around the time Hakan bastedt did that massive optimization work in the lw2 days... .

funinthefalls, 2016-09-05:
I understand that it will be an approximation, but that will give a rough idea of the total time the laser has been active.

krolco, 2016-09-05:
OK, I have it tracking the total accumulated job time, and it's output at the end of the job. Now I'm trying to figure out where to put it on the UI (along with a reset button). A bit longer.

funinthefalls, 2016-09-06:
The jog tab has quite a bit of free screen realestate. It might be possible to put it there, along with another enhancement request of mine, which would be having a machine name in the settings tab. A machine profile feature would also be nice, that way you could easily load different machines quickly in the LaserWeb software and switch between them. The machine profile could store info such as the cumulative job time. Personally, I have 3 different machines running laserweb, and I have to do the clunky job of using the saved json files for each machine to change between them. A drop down with each machine listed would be a god send.

krolco, 2016-09-06:
@funinthefalls I wound up adding a separate "Stats" panel, since the job time didn't seem to fit into anything else I was looking at. The new panel displays the elapsed time, and has a button to reset it. You can see it in #111 

As far as your enhancement request, maybe open a separate issue for it? The settings storage stuff isn't currently set up to save more than one set of settings to the browser, so it will require some significant rework, but I don't think it's impossible to do on this codebase. Might be something I can swing at.

krolco, 2016-09-06:
@yuusuf-sallahuddin You should be all set now.

funinthefalls, 2016-09-06:
Thanks for adding this.