LaserWeb/deprecated-LaserWeb3 #101: M5 undefined

sherzberg, 2016-08-28:
I've been running into this issue and digging around in the code, I don't see an `M5` variable set anywhere. Any help?

```
Missing error handler on `socket`.
ReferenceError: M5 is not defined
    at Socket.<anonymous> (/Users/sherzberg/workspace/LaserWeb3/server-grbl.js:126:18)
    at emitOne (events.js:96:13)
    at Socket.emit (events.js:188:7)
    at Socket.onevent (/Users/sherzberg/workspace/LaserWeb3/node_modules/socket.io/lib/socket.js:335:8)
    at Socket.onpacket (/Users/sherzberg/workspace/LaserWeb3/node_modules/socket.io/lib/socket.js:295:12)
    at Client.ondecoded (/Users/sherzberg/workspace/LaserWeb3/node_modules/socket.io/lib/client.js:193:14)
    at Decoder.Emitter.emit (/Users/sherzberg/workspace/LaserWeb3/node_modules/component-emitter/index.js:134:20)
    at Decoder.add (/Users/sherzberg/workspace/LaserWeb3/node_modules/socket.io-parser/index.js:247:12)
    at Client.ondata (/Users/sherzberg/workspace/LaserWeb3/node_modules/socket.io/lib/client.js:175:18)
    at emitOne (events.js:96:13)
    at Socket.emit (events.js:188:7)
```

funinthefalls, 2016-08-29:
M5 is set on the settings tab of LaserWeb (if you need to set it). Not all controller firmware requires the m3/M5 commands.

sherzberg, 2016-08-29:
I mean I don't see this being initialized anywhere in the code: https://github.com/openhardwarecoza/LaserWeb3/blob/master/server-grbl.js#L126

openhardwarecoza, 2016-08-29:
> L126

yeah that probably needs some "" around the m5

commit 7298f1a5 (2016-08-30): M5 should be sent as part of the message

fixes #101
closed 2016-08-30 by openhardwarecoza