LaserWeb/deprecated-LaserWeb3 #103: Proportional Feed rate adjustment

fumblz-n-action, 2016-08-30:
after some conversing with Ariel it has come to our attention that on the fly adjustment of the proportional feeds would make an awesome enhancement. being able to adjust while running could potentially aid in fine tuning each laser system in a minimal amount of time. this enhancement would also help generate less scrap material when working with new material

solarmill, 2016-08-30:
That's a brilliant idea.

openhardwarecoza, 2016-09-02:
note we need to get #106 in place first

gndprx, 2016-09-10:
I have this on my CNC and it's very useful. I was actually looking for this feature while testing today and noted that it would be a huge benefit.

openhardwarecoza, 2016-09-10:
#106.  (: needs to be done before this can start

openhardwarecoza, 2016-10-23:
Moved to wishlist on the readme https://github.com/openhardwarecoza/LaserWeb3/commit/b39c56a24a7de76afdbf95dd3e28ae12f6bf6c8c so I can close the issue for now

closed 2016-10-23 by openhardwarecoza