LaserWeb/deprecated-LaserWeb3 #106: Serial Enhancement:  Jump-the-queue for realtime feedrate commands, etc

openhardwarecoza, 2016-09-02:
I implemented https://github.com/openhardwarecoza/LaserWeb3/blob/master/server-smoothie.js#L263 in the backend (unshift puts the new gcode in the front of the queue array)

But its needs enhancement on the Frontend: If we are playing = true, then any commands snet should be sent as this instead so it lands in front of the queue

Right now, if you send say M220 overrides it lands BEHIND the job in the queue

openhardwarecoza, 2016-09-02:
PS: needs this in place before we can do #103

openhardwarecoza, 2016-10-23:
Moved to wishlist on the readme https://github.com/openhardwarecoza/LaserWeb3/commit/b39c56a24a7de76afdbf95dd3e28ae12f6bf6c8c so I can close the issue for now

cprezzi, 2016-10-29:
I started implementing overrides to the frontend. The idea is to extend the the DRO with displays for feed and speed overrides (% values) with buttons at side, to increase/decrease/reset the overrides.
Waht do you think about that?

openhardwarecoza, 2016-10-29:
Love it!  Go for it!  Sliders for overrides (:

cprezzi, 2016-10-29:
I think sliders are the wrong choice because of the way how overrides work in GRBL. GRBL has commands for increase or decrease by 1% or 10% and a reset command but no direct set of a % value. So my idea was to have a +, - and reset button aside a input-box which shows the actual % value. The override could be changed by either push one of the buttons or type in the value directly to the input box.

openhardwarecoza, 2016-10-29:
Ahh ok didn't know that.  Smoothie is M220 S[percentage]

cprezzi, 2016-10-31:
Good to know. I will try to implement smoothie version too, even if I don't have a smoothieboard (jet).

openhardwarecoza, 2016-10-31:
@arthurwolf (;

cprezzi, 2016-10-31:
Overrides for GRBL are working now, but I'm not perfectly happy jet. It's not very comfortable to only have +/- 1% buttons, but additional +/- 10% buttons would take to much space (see https://github.com/cprezzi/LaserWeb3/tree/Overrides) :)

[image]

@openhardwarecoza - Should I add a new "overrides" branch to your repo, or do you first want to test on my branch?

cojarbi, 2016-10-31:
Claudio

Maybe you could add a colapsable section bellow the  F and S K also you could please this on the same line

cprezzi, 2016-11-04:
What about pushing shift + button to get large stepps or stay on button for continous changes?

cprezzi, 2016-11-25:
I have implementet ctrl + button for large stepps and klick and hold for repeat.

openhardwarecoza, 2016-11-25:
If this is an seperate branch, feel free to bring it over to Master too.  The best feedback from our average user comes from "force the change on them and see if they rave or complain"

openhardwarecoza, 2016-11-27:
I merged it into Master... 

[image]

closed 2016-11-27 by openhardwarecoza