LaserWeb/deprecated-LaserWeb3 #108: Wish List

DarklyLabs, 2016-09-05:
As requested, here are some feature requests for using LW3 with the Emblaser 2.
I suspect some of these requests may already be handled in the CAM section, since they are common with cnc routers.

**Set material height**
Set the height of the material in the workspace for the correct laser head positioning.

**Laser Focal Length / Calibration Offset**
Value manually entered after calibration process has been run. This can either be a correction to a preset focal length or a corrected focal length value.

**Windows / Mac installer**
Easy installer for Windows and OSX.

**Easy Windows / Mac program execution**
Easy way to start LW3 within Windows and OSX.

**Air-Assist attached checkbox**
With the air-assist nozzle attached, this prevents z-height going too low and colliding with the material.
Without the nozzle attached, laser head can move 50mm.
With nozzle attached, laser head can move 10mm.

**Cutting Mat installed checkbox**
This adds a value to thickness of material based on the cutting mat thickness.

**WIFI**
LW3 to control a WIFI attached device.

**Camera to work on OSX**
Currently camera feature does not work on OSX.

openhardwarecoza, 2016-09-05:
Just to link issues together:  

Wifi support:  Waiting on https://github.com/luc-github/ESP3D/issues/112

funinthefalls, 2016-09-06:
Here is an easy short term solution to having an easy way to start LaserWeb within Windows until Peter can create a permanent elegant solution.

Enter the following text into Notepad

cd c:\laserweb3   (change this line to whatever the install path is)
start node server-smoothie.js
start chrome http://192.168.0.1:8000/

Save as  filename "LW.cmd" and file type as "all types"

I save this to my desktop, and when I want to start LaserWeb I just double click the LW.cmd file

DarklyLabs, 2016-09-06:
Thanks, that will certainly help.

funinthefalls, 2016-09-06:
For installing under Windows you can now use NPM install. It is documented at:
https://github.com/openhardwarecoza/LaserWeb3/wiki/Install:-Using-NMP-install-to-simplify-the-process

openhardwarecoza, 2016-09-07:
@funinthefalls  not really what Darkly needs.  We need to up to "dumb user" level so they can ship LW3 with the Emblaser2: https://darklylabs.com/emblaser2/

krolco, 2016-09-07:
@DarklyLabs are you interested in packaging with something like nw.js? It's a way to pack up a node app with an embedded Chromium browser... seems like it might fit the bill for packing up LW3 on Windows and Mac.

DarklyLabs, 2016-09-10:
@funinthefalls As Peter mentions we need an installation method for users who have very little computer experience. 
@krolco I'm not familiar with new.js but will link into it.

DarklyLabs, 2016-09-10:
@krolco @openhardwarecoza 
nw.js looks promising. I wonder whether it has ever been considered for distributing LW?
There is documentation on creating a distribution package here: http://docs.nwjs.io/en/v0.13.0/For%20Users/Package%20and%20Distribute/

Would anyone be able to write a basic code skeleton for us to start testing LW installation on Windows and OSX?

krolco, 2016-09-10:
This requires a bit of study, because I'm not quite sure how the serial comms work in this structure. That said, I think it's promising because it's a way to package web apps that should be click-and-run.

krolco, 2016-09-11:
For fun, I just tried dumping all the HTML/JS/CSS into an nw.js project, just to see how much would work. Comms are broken as expected, but so is a lot of other functionality. This will for sure be a project to look into packaging on this; would be cool if it works!

openhardwarecoza, 2016-09-11:
nw,js was investigated rather extensively in Jun 2015 - although in LW1 the suggestion to use Electron.io https://github.com/openhardwarecoza/LaserWeb/issues/21 also popped up - both seemed to have the bigger issue in that the node-serialport module has to be compiled for the specific OS. 

There was a very good suggestion in that thread though: To use the Chrome Serial API https://developer.chrome.com/apps/serial - which would be even better.  "packaging" can then be replaced with "install from chrome extensions"

openhardwarecoza, 2016-09-11:
See https://www.npmjs.com/package/browser-serialport

krolco, 2016-09-11:
Hmm, looks like I'm covering ground that has already been covered. It does look like Electron maps a little better to the problem than nw.js does, but there's still stuff to work through.

openhardwarecoza, 2016-09-12:
*Nudge* - I spoke to Domenic this morning and since they want to ship machines in Dec, this is becoming stressful

[9:32:20 AM] Domenic DiGiorgio: How can we help with things? Unfortunately we can’t program so we are useless in that area?
[9:32:44 AM] peter.van.der.walt.netcb: I'll ask the other contributors if they need any help.

So, how can they help us, get this going.  a few of the things can perhaps form part of the much asked for machine profiles: 
Set material height, Air-Assist attached checkbox, Laser Focal Length / Calibration Offset (Z moves?), Cutting Mat installed checkbox

krolco, 2016-09-12:
@openhardwarecoza @DarklyLabs I saw these requests; these seem straightforward to add to the settings panel (which will eventually become profiles), but I'm not sure what to do with the data -- I don't know if this is supposed to change the GCode generate, or emit specific commands to the laser, or what. Is there more info somewhere on how these get hooked up to the hardware? These don't seem hard; I just don't know how they are supposed to interact. For instance: for the cutting mat checkbox, is that something that we pass along as an M command, or is that something that is supposed to offset the Z height of the laser? Is the cutting mat thickness standardized? We don't currently have a setting for mat thickness that I saw. 

Anyway, if there is a bit more spec on what affect these are supposed to have, I might be able to swing at them.

krolco, 2016-09-14:
I'm going to split this list into some separate requests, so they can get tracked separately. I'll close this one once it's split.

closed 2016-09-14 by krolco