LaserWeb/deprecated-LaserWeb3 #109: Error when opening gcode file in Laserweb

funinthefalls, 2016-09-05:
If I open an existing gcode file in LaserWeb, the following error is displayed in the terminal:

ReferenceError: boundingBox is not defined (http://192.168.0.101:8000/js/tabs.js on line 97)

The file seems to load ok however, it shows up in the all layers tab, but it does not display anything in ther Gcode view tab.

openhardwarecoza, 2016-09-05:
> tabs.js on line 97)

Since a gcode file is but a layer.  Click generate gcode on cam tab to
populate the 'gcode tab'  and prep job for sending.

funinthefalls, 2016-09-06:
Still does not populate the gcode tab after clicking generate gcode on the cam tab. And in the Gcode tab on the left (not to confuse it with the gcode tab at the top) It has a "Start Gcode" box, a "Gcode loaded from file" box, and a "gcode end" box.

openhardwarecoza, 2016-10-23:
Duplicate of https://github.com/openhardwarecoza/LaserWeb3/issues/163

closed 2016-10-23 by openhardwarecoza