LaserWeb/deprecated-LaserWeb3 #113: Settings check is never satisfied.

MacTheApple, 2016-09-07:
Aafter filling out all the "required" settings, pressing save, and refreshing leaves me with the " Incomplete Config Please complete configuration" banner at the top of the settings page and the  ": MISSING CONFIG: You need to configure LaserWeb for your setup. . Click Settings  on the left, and work through all the options"
banner in the console. 
The only way to make these banners go away is to fill out all the commands, even the ones that are  "optional" , which makes them seem not really that optional at all.

krolco, 2016-09-07:
Which browser are you using?

krolco, 2016-09-07:
Just to double check what we are debugging, are you using the latest version of the code? I'll assume you are.

I looked at the code, and the only code path that should cause the banner to appear is if you are missing a required setting. Are there any log lines in the console on the page in red that say "Missing required setting:"? If you can post the log output from startup, that would be helpful also.

MacTheApple, 2016-09-07:
I'm using the current version of chrome.
The 2 settings that are shown as optional in the interface, but are required in the systemlog are "start gcode" and "end gcode"  I have been solving this by just putting an M5 in each spot.
in systemlog:
Missing required setting: endgcode
Missing required setting: startgcode
in settings:
Start G-Code (Optional)
End G-Code (Optional)

openhardwarecoza, 2016-09-07:
A recent bug was introduced when localstorage.js was refactored,  ignored
it for now...

openhardwarecoza, 2016-09-07:
For those who want to bugfix: https://github.com/openhardwarecoza/LaserWeb3/blob/master/public/js/localstorage.js#L16    (after refactored from what we used to have (from older commit:  https://github.com/openhardwarecoza/LaserWeb3/blob/dd71d92be663b39b5bb567a5fba550e99e8ec903/public/js/localstorage.js#L61  - see how I ignored some params  (;)

@lautr3k 's refactor broke that a little

krolco, 2016-09-07:
I just submitted #118 for this.

krolco, 2016-09-07:
@MacTheApple can you please test this now? Thanks!

MacTheApple, 2016-09-08:
Tested and working, 
Thank you.

closed 2016-09-08 by MacTheApple

gfetters, 2016-11-12:
This problem is back.  I can not save settings even if I fill out all settings.  I am running on a raspberry pi 3.  latest build

openhardwarecoza, 2016-11-12:
Screenshot of the syslog where it does the check?

funinthefalls, 2016-11-12:
I just did a git pull and loaded LaserWeb. I removed any setting that was optional and refreshed the browser. The software behaved as expected, no configuration errors in the syslog. I am running Windows 10, and latest version of Chrome. LaserWeb version 30420. Tried various combinations of empty optional settings, still not able to reproduce.