LaserWeb/deprecated-LaserWeb3 #114: object to default parameters when importing new object

fumblz-n-action, 2016-09-07:
I imported an image and sized it accordingly and gave it an off set. next i imported another object (dxf file) then image file reverted back to its defaulted original imported size and placement,

krolco, 2016-09-07:
I am able to reproduce this. Someone (maybe me?) will have a look.

krolco, 2016-09-07:
OK, I've been looking at the code, and I think is a side-effect of an earlier feature that I did ( #104).
I have to think about how to fix this a bit more. I'll try to look at it again tonight.

krolco, 2016-09-08:
Ok, I just merged #120 that should finally fix this -- I didn't quite get it right the first time, but I spent more time grokking the code, and it should work now. I was accidentally resetting the raster's position anytime another object was added to the scene (because I didn't understand that I added the code in a spot that made a pass over all objects every time something was added to the scene, instead of once at first add). @fumblz-n-action, please test this now if you can.

cojarbi, 2016-09-08:
I can confirm

krolco, 2016-09-09:
Does this seem OK now? I'd like to close it if possible. @cojarbi @fumblz-n-action Thank you!

cojarbi, 2016-09-09:
I did check it and passed fine

AY

krolco, 2016-09-09:
Thank you. I'm going to close this one then.

closed 2016-09-09 by krolco