LaserWeb/deprecated-LaserWeb3 #121: Deleting a raster layer will not delete image "preview" on grid

cojarbi, 2016-09-08:
So this could be a side effect but:
open a raster then close that tab, the image will stay on the grid until another tab or the all layers tab is pressed. This does not happens with vectors

Not the solution but i will say a good practice would be that every time a tad is closed either select the previous one or maybe better select the all layers tab

http://g.recordit.co/K94i9fXJwk.gif

funinthefalls, 2016-09-09:
Was able to duplicate, however, the after deleting the raster, it still appears in the All layers tab.

cojarbi, 2016-09-09:
Appears the generated gcode or the original raster? 

AY

cojarbi, 2016-09-12:
Here is another run. The generared preview of the gcode appears on the All layers tab but not on the gcode, it should not appear at all
http://recordit.co/qrxS6doPOs

krolco, 2016-10-10:
More info from @funinthefalls:

Steps to reproduce:
1. Open any raster or vector image
2. Generate Gcode
3. Delete opened image tab using the little X
4. Generated Gcode tab does not have a little X to delete it. Only way to delete Gcode and work with a different image is to refresh the browser, which disconnects the smoothieboard temporarily.
SHouldn't there be a way to delete the generated Gcode in case you realize you have used wrong fees rates or some other setting and want to regenerate the Gcode with different settings?

funinthefalls, 2016-10-10:
That is not what I had happen, well that part did, but additionally, on the left hand pane where the actual Gcode is, you can not delete the gcode if you want to start over, you have to refresh the browser. Does that make more sense now?

openhardwarecoza, 2016-10-10:
well, until you regerenate gcode, what you see is actually true state:  That "processed gcode" is still present on the gcode tab...

funinthefalls, 2016-10-11:
OK, I undersdand now, Thanks Peter

closed 2016-10-23 by openhardwarecoza