LaserWeb/deprecated-LaserWeb3 #122: Add indicator while processing trace from vector

cojarbi, 2016-09-09:
Currently there is not indication that there an ongoing process. For large or complex images trace and take some time. Maybe add the same indicator as raster

krolco, 2016-09-09:
I just added #125 for this. Give it a whirl.

cojarbi, 2016-09-09:
I do see that something its going on as now the screen goes darker, but the message comes and goes to fast. Try it with the attached image.

If you can please use the same loading indicator as the raster process. Besides not being intrusive it also lets you do some other stuff in the screen

[image]

openhardwarecoza, 2016-09-09:
http://ricostacruz.com/nprogress/ .start, .inc, and done should do...

krolco, 2016-09-09:
Ah, I didn't notice that there was an existing progress indicator for the UI. I'll change over to that

krolco, 2016-09-09:
@openhardwarecoza So, Potrace doesn't seems to fire any progress events, so I don't think anything will call .inc at all, but that's OK, because nprogress fakes progress automatically :) I'm working on this one; I'm out of town for a bit, but I'll get it.

krolco, 2016-09-10:
Messed with this some more... Potrace runs the Javascript thread at 100% while it's tracing, and it seems that no progress events fire, even when I put it on a dedicated timer, and even when I stuck an .inc() call directly in Potrace. I'll have to work on this one more.

krolco, 2016-09-14:
So I'm stuck on this simple request; the JS just wedges up tight while Potrace is running, and I don't get any motion of the bar. I'm tabling this for a bit until I think of something else.

openhardwarecoza, 2016-09-15:
We may need to rewrite the potrace library to work in chunks like I did here https://github.com/openhardwarecoza/LaserWeb/commit/41ee83ed3e7124e4cf346e2cf78de991648e818d#diff-681a90cbc63da7ac6566acc75887b718R203 

that doChunk function takes what would normally be a loop, and does it in chunks instead - each chunk ties up the browser for maxTimePerChunk,  after which it releases, we can then do a GUI update like nprogress, and then it takes the next chunk for processing...

openhardwarecoza, 2016-09-15:
https://github.com/openhardwarecoza/LaserWeb3/blob/74feb7fe91b1084f7062b398a309f3792f3fa2e1/public/lib/gcode-viewer/gcode-parser.js#L168 also uses doChunk

krolco, 2016-09-15:
Oh boy... I'll have to spend some time to grok that one. For now, I did change it over to NProgress, although it doesn't update much if at all while Potrace is running. It's at least consistent with raster generation now, though.

openhardwarecoza, 2016-10-23:
Closing this for now...

closed 2016-10-23 by openhardwarecoza