LaserWeb/deprecated-LaserWeb3 #123: Allow user to reset grbl through laserweb?

MacTheApple, 2016-09-09:
If you are using grbl with soft limits enabled, and you try to go out of bounds, you need to be able to send a control-X to grbl to get it to start accepting commands again, but this doesn't work through laserweb. From looking at grbl's source code, I can see this:
#define CMD_RESET 0x18 // ctrl-x.

So would it be possible to make a reset button on the interface that sends `&#x18;` to the serial port when pressed? I'm not near my laser right now, but next time I can, I will test out setting that as a macro if It will accept it. Even if it works, it would still be nice to have that on the main control area instead of hidden away in the macro panel.

openhardwarecoza, 2016-09-09:
Macro button on the macro panel should do...

openhardwarecoza, 2016-09-09:
Regarding putting it in main UI:  The rule stands. Firmware specific things make maintenance hell

MacTheApple, 2016-09-09:
It doesn't work as a macro anyway, the commands get stuck in the queue because when grbl is in a soft alarm state, it doesn't respond to ?. By the way, laserweb3 doesn't detect this as an alarm state. (still shows idle at top of screen)
: ALARM: Soft limit
: [Reset to continue]
in order for a  macro to work here, there would need to be some way to ignore the fact that the machine is not responding.

openhardwarecoza, 2016-09-09:
Thats where https://github.com/openhardwarecoza/LaserWeb3/issues/106 is needed...

closed 2016-09-09 by openhardwarecoza

MacTheApple, 2016-09-09:
Maybe have a look at https://github.com/cheton/cnc/blob/master/src/app/controllers/Grbl/GrblController.js
I wonder if it would be possible to do a merge of the laser specific functions of laserweb into the controller code from there? combine resources? The code over there looks similar, but has very limited laser functionality.