LaserWeb/deprecated-LaserWeb3 #134: Laser Focal Length / Calibration Offset

krolco, 2016-09-14:
Value manually entered after calibration process has been run. This can either be a correction to a preset focal length or a corrected focal length value.

krolco, 2016-09-15:
Total Laser Focal Length, Material Thickness, and Cutting Mat Thickness are all on the UI.

I have it hooked up to raster generation, but not to SVG/DXF generation yet (coming soon). This stuff is hooked up in the emblaser branch.

krolco, 2016-11-30:
This should be coded now. Please test.

openhardwarecoza, 2016-12-02:
@krolco maybe I'm being blonde but I dont see the commit for this?

krolco, 2016-12-02:
It is (was?) the Z-Focus height param on the Size settings panel -- should be 0 for most cases, but according to @DarklyLabs can contain an offset for calibration for a specific machine. It is in the source code I'm looking at, but I'll make sure tonight that it it didn't get damaged during any recent merges.

krolco, 2016-12-02:
You can see some work on it here: https://github.com/LaserWeb/LaserWeb3/commit/d9b7528dc9474862f9f7063b2ce32f0f5f5ba2d9

openhardwarecoza, 2017-03-05:
Closing - wont be fixing LW3 anymore

closed 2017-03-05 by openhardwarecoza