LaserWeb/deprecated-LaserWeb3 #135: Windows/Mac Installer and Easy Start

krolco, 2016-09-14:
Windows / Mac installer
Easy installer for Windows and OSX.

Easy Windows / Mac program execution
Easy way to start LW3 within Windows and OSX.

krolco, 2016-09-14:
See also #108.

From that thread:
 @funinthefalls:

Here is an easy short term solution to having an easy way to start LaserWeb within Windows until Peter can create a permanent elegant solution.

Enter the following text into Notepad

cd c:\laserweb3 (change this line to whatever the install path is)
start node server-smoothie.js
start chrome http://192.168.0.1:8000/

Save as filename "LW.cmd" and file type as "all types"

I save this to my desktop, and when I want to start LaserWeb I just double click the LW.cmd file

For installing under Windows you can now use NPM install. It is documented at:
https://github.com/openhardwarecoza/LaserWeb3/wiki/Install:-Using-NMP-install-to-simplify-the-process

From @openhardwarecoza:

See https://www.npmjs.com/package/browser-serialport

copet80, 2016-09-30:
What about packaging and delivering the app with Electron? Overkill?

openhardwarecoza, 2016-09-30:
Node serialport and Electron doesnt play too nicely together,  or so I have
been told. That said,  do you feel up to giving it a go for us? (:

copet80, 2016-09-30:
Sure that'd be worth trying. What'd be the best way to test the serialport? Connect it to Arduino? Or is it better to test with a LaserWeb compatible hardware?

openhardwarecoza, 2016-09-30:
You can test an arduino running Grbl (:

copet80, 2016-09-30:
No problem. I assume this library suffices?
http://blog.protoneer.co.nz/grbl-arduino-library/

openhardwarecoza, 2016-09-30:
Yip,  or the precompiled hex uploaded using xloader or avrdude.  Either is
fine.

copet80, 2016-10-01:
One thing to consider pretty soon is the certificate signing for Mac OSX. The below article is a bit old but should give an indicator on the process involved. 
https://successfulsoftware.net/2012/08/30/how-to-sign-your-mac-os-x-app-for-gatekeeper/

Having said this, have you got any prior experience on this?

pabloevaristo, 2016-11-28:
looking for a good controller for my chinese 2.5W engraver, with arduino nano and grbl installed.
I have followed the wiki "Install: Using NPM install to simplify the process" (in windows 7 x64) but got a couple of warnings and can´t start the app.
I attached the output from the console.
Superthanks in advance﻿
[image]
[image]

openhardwarecoza, 2016-11-30:
Closing this issue... Seems the electron branch works nicely - we'll address the individual problems in #196 #195 #194

closed 2016-11-30 by openhardwarecoza