LaserWeb/deprecated-LaserWeb3 #138: WIFI support

krolco, 2016-09-14:
LW3 to control a WIFI attached device.

krolco, 2016-09-14:
From #108:

Wifi support: Waiting on luc-github/ESP3D#112

openhardwarecoza, 2016-10-11:
I forked and started working on a ESP8266 firmware here https://github.com/openhardwarecoza/LaserWebEsp8266

openhardwarecoza, 2016-10-11:
Pushed some very early code into LW too https://github.com/openhardwarecoza/LaserWeb3/commit/59e12e2cb9af1571573ec62e555a30e07ee04c34

openhardwarecoza, 2016-10-11:
@arthurwolf @wolfmanjm - what is the strategy to upload file over UART?

M28 file.gcode
then, send each line:  Should I wait for an OK between lines?
once all sent, then M29?

arthurwolf, 2016-10-11:
I don't have the exact protocol in mind, but I think the reference
implementation of it you want to be following, is the one in pronterface.
You definitely want to wait for Ok yes.
It's insanely slow though ...

openhardwarecoza, 2016-10-11:
Ok,  got it going with waiting for 'ok'  between lines.
As expected,  it is darn slow.

The other major stumbling block,  is that during sd upload,  we send
faster,  than the TCP buffer on the ESP8266 can flush.  So i can send about
16kb of data,  then i need to pause a few seconds,  till the free memory
heap on the esp recovers back to say 30kb free.  If the head drops below
10k free,  the watchdog reboots the esp :(

@darklylabs i must say,  i dont think we'll manage SD uploads at all.
Maybe Sneakernet (pull sd,  use card reader to copy file,  place sd back in
machine)  -  wifi will have to be restricted to Jog and setup moves,  and
starting jobs already on SD,  monitoring progress and pausing stopping
actions.  But at this speed,  walking over to the machine to get the SD is
much much faster :( -  raster jobs literally take hours to upload

> @arthurwolf <https://github.com/arthurwolf> @wolfmanjm
> <https://github.com/wolfmanjm> - what is the strategy to upload file over
> UART?
>
> M28 file.gcode
> then, send each line: Should I wait for an OK between lines?
> once all sent, then M29?

> <https://github.com/openhardwarecoza/LaserWeb3/issues/138#issuecomment-
252867410>,
auth/AAGpFd-pDjpJ3kzTwnrEXXDd-5EXbcbiks5qy19vgaJpZM4J8bQD>

openhardwarecoza, 2016-10-11:
Not a surprise, but streaming jobs, also a No Go. Vector sort of works, but as can be seen in the test below, goes haywire around 60% of the way through when the heap drops.   Streaming a raster job, less than 1 second of engaving later, goes bonkers.

@darklylabs Domenic, I think its time for us to discuss a "what was promised" in terms of Wifi? 
We can stick a simple "monitoring" dongle on the ESP and use Wifi (phone/tablet) as a simple UI for driving the machine. No CAM, no Upload, no dealing with the files... Thats something I was working on for Mark Carew (OpenBuilds) already - to make the mobile device a handheld MPG. Would that strategy suffice?

[image]
[image]

openhardwarecoza, 2016-10-11:
Giving it one last test using https://github.com/Links2004/arduinoWebSockets/issues/61#issuecomment-252689452

openhardwarecoza, 2016-10-11:
Hold up!  @Links2004 sketch seems a whole million better!  I am busy
streaming a raster job at the moment.  Hold thumbs it finishes without a
crash but so far so good!

openhardwarecoza, 2016-10-11:
[image]

Eureka!  That raster and the correct vector was both streamed over wifi! @darklylabs

openhardwarecoza, 2016-10-11:
Almost done:  See https://plus.google.com/u/0/+PetervanderWalt/posts/7QT3xGRECSU

Todo:  Add a Play Button (:

wolfmanjm, 2016-10-11:
FYI you can also upload ascii files using the following protocol...

issue an upload /sd/filename.g 
then stream the ascii data, and send a control Z when the file is done.
Do not try this while printing though as is disables normal serial communication.
you can use the builtin md5sum /sd/filename to check the file and do the same on the host.

openhardwarecoza, 2016-10-11:
Thanks Jim! I'll try that in the morning (:

closed 2016-11-27 by openhardwarecoza