LaserWeb/deprecated-LaserWeb3 #139: Camera to work on OSX

krolco, 2016-09-14:
Currently camera feature does not work on OSX

DarklyLabs, 2016-09-22:
Something to note is that there doesn't appear be a way to select which camera to use on systems with multiple cameras.

This is important when working on a laptop with an inbuilt camera and an external USB camera attached.

copet80, 2016-09-30:
Which feature does not work on OSX? Which browser?

openhardwarecoza, 2016-09-30:
WebRTC needs https to work. The nodejs localhost is served as http.

openhardwarecoza, 2016-09-30:
It does print the error in the console log,  if you enable Video overlay,
then refresh the browser. .

openhardwarecoza, 2016-10-06:
Affects all OSs after recent Chrome updates

openhardwarecoza, 2016-11-27:
@copet80 any update on solutions you came up with?

lautr3k, 2016-11-27:
Fixed on optimization branch

openhardwarecoza, 2016-11-30:
Resolved in the Optimisation branch by @lautr3k

closed 2016-11-30 by openhardwarecoza