LaserWeb/deprecated-LaserWeb3 #144: Optimize cut path multiple passes on open vector

cojarbi, 2016-09-21:
When cutting an open vector in multiple passes, for example a line in 2 passes, it does 2 left to right passes instead of running one left-right and then one right-left

krolco, 2016-09-22:
Peter and I were chatting about adding a path optimizer to LW3 -- I think it might catch this kind of bad behavior. Hopefully something we can get to!

funinthefalls, 2016-10-17:
https://hackaday.io/project/4955-g-code-optimization  Maybe this can be used?

jorgerobles, 2016-10-17:
Better! http://parano.github.io/GeneticAlgorithm-TSP/

openhardwarecoza, 2016-10-23:
@cojarbi I dont know of any optimisers that would reverse lines to optimise it. And for CNC that would be totally wrong too (converting a climb cut to a conventional cut in the name of optimising speed...)
The bigger secret here is properly closed vectors (polyline dxf for the win)

That said, still happy to accept a pull for adding something to make less G0 moves (:

openhardwarecoza, 2016-10-23:
Moved to wishlist on the readme https://github.com/openhardwarecoza/LaserWeb3/commit/b39c56a24a7de76afdbf95dd3e28ae12f6bf6c8c so I can close the issue for now

closed 2016-10-23 by openhardwarecoza