LaserWeb/deprecated-LaserWeb3 #146: Camera under windows

DarklyLabs, 2016-09-23:
I have found that when working under windows, connecting to LW3 via Localhost:8000 allows the camera overlay to work as expected.

Connecting via http:/xxx.xxx.xxx.xxx:8000 does not allow the camera to be found by LW3.

I'm running Win10 and a standard USB2.0 camera.

Is this a known issue?

DamianTrejtowicz, 2016-09-24:
my camera is working until i load any file, then picture gone,
how to use this feature?

openhardwarecoza, 2016-09-24:
Camera at this point is a dead feature...  If you update,  the new version
removes webrtc support,  and instead allows ypu to specify a camera URL.
Here you can pick whatever streaming service you want to use..
Mjpg_streamer on pi/linux for example

DamianTrejtowicz, 2016-09-24:
shame, looks like i need to buy some ip camera if i want use this feature or setup another pc with webcam
what linux is the best for pc with intel atom ?

openhardwarecoza, 2016-09-24:
Just install ANY webcam to web app,  like Mjpg_streamer...  Ip camera
itself will probably not work unless it gives an jpg stream

openhardwarecoza, 2016-09-24:
Also,  all written up,  in the Wiki
https://github.com/openhardwarecoza/LaserWeb3/wiki/Settings:-Remote-Webcam
as always

DarklyLabs, 2016-09-25:
How did we miss the fact camera was killed off?
The E2 has an integrated USB camera and this is one of the key features were were relying on with LW.
How can we get this feature back in?
Is there a work around for a USB camera across OSX & Win?

openhardwarecoza, 2016-09-25:
Not yet.  Webrtc and localhost doesnt play so well.  We'll need to host
webcam from the nodejs server too

openhardwarecoza, 2016-10-06:
Duplicate of #139 -  closing

closed 2016-10-06 by openhardwarecoza