LaserWeb/deprecated-LaserWeb3 #147: Scientific notation in generated GCODE generates error in GRBL

maudette01432, 2016-09-25:
Scientific notation in generated GCODE generates an error in GRBL.  This in turn causes the LW3 communications to lockup and in my case leaves the laser in an unsafe state (stalled and still firing laser).

The console server shows the following:

Sent: G1 F1200 X9.02687276038705e-7 Y18.46436716475577 Z0 S255 Q: 2575
Recv: error: Unsupported command

The only way to get the laser into a safe state is to reset the controller again.  Universal GCODE Sender (UGS) continues on it's way even after the error resulting in an engrave (likely with a missed coordinate).  This GCODE may also be causing the issues with my RAMPS setup locking up or loosing communication as well.  

Since the precision on my banggood A3 2.5W is only about 0.0125 mm step (80 steps/mm)...this type of precision is overkill (likely on any laser setup).  It would be best to output standard fixed notation with 9 units of precision following the decimal (if in metric/mm).  Even in inches this should be sufficient for accuracy.

There are potentially 2 issues here:
1) Generated GCODE should used fixed precision
     1a) The precision should be one ten-thousandth of an inch (0.0001"), which is a larger distance than the usual minimum increment in G21 (one thousandth of a millimeter, .001) 
     1b) 1a is referenced from the G20/G21 documentation on ghe G-code wiki
2) When Gcode generates an error condition the sender needs to be able to recover

emteeoh, 2016-09-27:
um... Was this GCODE generated by LW? if so, what was it generated from?

maudette01432, 2016-09-27:
This code was generated from LaserWeb3.  Imported SVG scaled at 120 DPI and Exported with my machine settings.  I have no START GCODE and no HOMING Sequence so there is nothing at the start of the file (including a header - maybe a but because the "Required" Homing sequence field was left empty?)

Windows 10 grbl-server.

I'll upload my config file today.  100% this was generated with LW3 and I can reproduce it with the SVG file.

openhardwarecoza, 2016-09-27:
I suspect we need a toFixed(2) somewhere there in threegcode.js

openhardwarecoza, 2016-10-01:
Can you upload the file that gave you that issue,  so I can test while fixing?

maudette01432, 2016-10-01:
[sf_hop_test.zip](https://github.com/openhardwarecoza/LaserWeb3/files/504332/sf_hop_test.zip)

I've attached a modified version of the file here...it does however still reproduce the problem (it's in the outline - I just removed some of the "core" junk.

Open SVG sf_hop_test.svg
On the SVG set DPI to 120
Press Generate G-Code
Save Gcode

I've also attached the config I'm using for my Banggood A3 2.5W engraver (JT-Photonics GRBL 0.9)
[A3_2.5W_LASER.json.zip](https://github.com/openhardwarecoza/LaserWeb3/files/504334/A3_2.5W_LASER.json.zip)

maudette01432, 2016-10-03:
I modified my local copy of threegcode.js and formatted all the output strings using xpos, ypos, and zpos to utilized .toFixed(4) - this still has accuracy required for G20/G21 reduces output file sizes and fixes the issue with the scientific notation.  I didn't clone the repo, but will if you want me to commit the change.

I have attached the modified file here.

[threegcode.zip](https://github.com/openhardwarecoza/LaserWeb3/files/507143/threegcode.zip)

gndprx, 2016-10-05:
What is considered acceptable resolution? 4-digit should be beyond accurate for the capabilities of most machines I would think.

maudette01432, 2016-10-05:
As mentioned earlier.  4 digits covers us for G20 and G2 (inches and metric).

1a) The precision should be one ten-thousandth of an inch (0.0001"), which is a larger distance than the usual minimum increment in G21 (one thousandth of a millimeter, .001) 
1b) 1a is referenced from the G20/G21 documentation on the G-code wiki

openhardwarecoza, 2016-10-05:
Please send it as a pull request: easier for me to merge (i'm on the road -
taking a bit of a quick holiday) - i can merge the pull via github's
webpage (:

openhardwarecoza, 2016-10-06:
I gave @maudette01432 push access to both lw3 and lw4 repos

maudette01432, 2016-10-06:
Tried to push...getting permission error

fatal: unable to access 'https://github.com/openhardwarecoza/LaserWeb3.git/': The requested URL returned error: 403
root@LENOVO-LAPTOP:/mnt/c/LW3_EDIT/LaserWeb3/public/js# git push origin master
Username for 'https://github.com': XXXX
Password for 'https://XXXX@github.com':
remote: Permission to openhardwarecoza/LaserWeb3.git denied to maudette01432.
fatal: unable to access 'https://github.com/openhardwarecoza/LaserWeb3.git/': The requested URL returned error: 403

It's letting me login with said username/password (I changed XXXX to protect me).   Do I have to fork before I can push and have someone else merge in the change.  (Github noob..have used git for other projects but not with github).

openhardwarecoza, 2016-10-07:
> git/': The requested URL returned error: 403

Github shows you havent accepted the invite yet

maudette01432, 2016-10-07:
Thanks Anthony.  Normally I just use git locally.  Accepting the invite and I was able to push.

closed 2016-10-07 by maudette01432

openhardwarecoza, 2016-10-07:
Thanks for the help!