LaserWeb/deprecated-LaserWeb3 #148: Send arcs instead of tiny line pieces for arcs and circles

McNugget6750, 2016-10-05:
I'm using GRBL 0.9j with a custom board but nothing special, all within GRBL specs.
I use a K40 laser, pin D11 for hardware PWM and pin D13 for inverted laser enable.

Whenever I laser a curve LW3 sends thousands of little vectors to the GRBL controller who is hopelessly underpowered to deal with this type of look-ahead. It gets worst when the radii get very small.

Currently, there is no way of setting the granularity of those vectors.
Also there is no way of using arc commands in LW3 instead of tiny line commands.

Arc commands would significantly improve cutting speed and efficiency with buffers and data communication.

openhardwarecoza, 2016-10-06:
Pull requests gladly accepted.  Look at js/threegcode.js

openhardwarecoza, 2016-10-06:
Nb note from Todd, but we can never reliably calculate offset paths as used in the CNC cam portion, from Arcs. Lines are much safer. . Investigate your Grbl Acceleration

closed 2016-10-06 by openhardwarecoza