LaserWeb/deprecated-LaserWeb3 #149: Implement path optimizer for cutting DXF files

McNugget6750, 2016-10-05:
When importing any DXF from solidworks or most other dxf exporting tools, the individual objects are separated and in random order and direction. Some tools like dxf2gcode implement a "traveling salesman" algorithm to optimize tool paths and to bring all vectors and arcs into one single shape with a common traveling direction (CW or CCW).

Additionally, please implement sending arcs in the following format. Again, dxg2gcode implements an arc detection algorithm using a biarc Fitting algorithm. Maybe there is a way to integrate that?
G2 X -36.061 Y -19.061 I  -1.061 J  -1.061

openhardwarecoza, 2016-10-06:
Pull requests gladly accepted: take a go at coding. It (:

openhardwarecoza, 2016-10-06:
Nb note from Todd,  but we can never reliably calculate offset paths as used in the CNC cam portion,  from Arcs.  Lines are much safer. . Investigate your Grbl Acceleration

closed 2016-10-06 by openhardwarecoza