LaserWeb/deprecated-LaserWeb3 #153: Can not clear Gcode once generated.

funinthefalls, 2016-10-10:
Steps to reproduce:
1. Open any raster or vector image
2. Generate Gcode
3. Delete opened image tab using the little X
4. Generated Gcode tab does not have a little X to delete it. Only way to delete Gcode and work with a different image is to refresh the browser, which disconnects the smoothieboard temporarily.
SHouldn't there be a way to delete the generated Gcode in case you realize you have used wrong fees rates or some other setting and want to regenerate the Gcode with different settings?

cojarbi, 2016-10-10:
@funinthefalls this is the same as we discussed on #121 
Just that there i did not mention the gcode.

I agree with you, in the meantime just load another file and process it, this will delete the previews gcode and preview, no need to refresh the browser

krolco, 2016-10-10:
I'm going to dupe this to #121, so we don't have double work on it. @funinthefalls, I'll add your comment to the thread on #121.

closed 2016-10-10 by krolco