LaserWeb/deprecated-LaserWeb3 #161: [Enhancement] Refactor threegcode on to a library

jorgerobles, 2016-10-18:
Issue #160 comes for a forth study of refactoring the threegcode on to a library, allowing to use gcode generator alone, decoupling from interface dependencies (like laserMultiply fetching from UI field) and doing "dependecy injection" of application settings and maybe document (SVG, DXF) scoped settings.
This will allow to improve #52 development.

openhardwarecoza, 2016-10-18:
Agreed.  Also @tbfleming is helping us add his cam code from jscut.org,  so
definitely the cam portion needs a lot of work.  Again,  this needs to
happen in the step of porting lw3 sections to lw4.  Speak to @lautr3k and
see if he can give some guidance on how best to start porting code over to
lw4?

jorgerobles, 2016-10-18:
Perfect!

openhardwarecoza, 2016-10-23:
Can I close this issue, and we continue looking at it in the react/redux based LW4?

closed 2016-10-23 by openhardwarecoza

jorgerobles, 2016-10-25:
Of course!