LaserWeb/deprecated-LaserWeb3 #162: [Proposal] MVC like pattern for LaserWeb

jorgerobles, 2016-10-18:
Following #161, I want to raise a proposal for LS3~LS4 decoupling code, namely separating UI functions from core application, forth splitting on libraries, etc., so the code could be improved better from different contributors and aspects.
Maybe it could be ported to Angular/Ember frameworks in a future

openhardwarecoza, 2016-10-18:
I totally agree,  have a look at the lw4 codebase: switching to ReactJS

jorgerobles, 2016-10-18:
Ow Yeah!

openhardwarecoza, 2016-10-18:
Which @tbfleming is addressing already with React
https://plus.google.com/101442607030198502072/posts/VzfbPqr26Ye

openhardwarecoza, 2016-10-18:
PS: I gave you push access to LW4 a while ago, but I see the invitation is still unaccepted

https://github.com/openhardwarecoza/LaserWeb4/invitations

jorgerobles, 2016-10-18:
Found! I missed that. Thanks!

openhardwarecoza, 2016-10-18:
All cool! Anyone else who wants push access, just ask (:  - this project is way bigger than me (:

closed 2016-10-18 by openhardwarecoza