LaserWeb/deprecated-LaserWeb3 #163: Gcode doesn't load no error thrown

maudette01432, 2016-10-19:
This GCODE is output from Fusion360's LASER/PLASMA/WATERJET CAM.  Loads just fine in UGS but wil not load in Laserweb3.  No Error output is given to console.

[DONT_PANIC.zip](https://github.com/openhardwarecoza/LaserWeb3/files/539889/DONT_PANIC.zip)

jorgerobles, 2016-10-19:
Do you mind if you share the postprocessor? Thanks!

maudette01432, 2016-10-20:
I will post something up on my blog later today.  The general gist is I started with the built-in smoothie post and modified it slightly to more closely support the features of my Banggood A3 engraver. (Taking out unsupported commands) There's currently a number of things that I don't like - but GCODE generation seems to work ok and run just fine not he laser w/ (GRBL 0.9g JTECH) - I haven't made the jump to the 1.1c which includes the new spindle code.

openhardwarecoza, 2016-10-23:
The 3D viewer needs the S param in line with the G1's 

From LaserWeb
`G1 X59.200 S0.01 F1200
G1 X59.000 S0.02 F1200
G1 X58.400 S0.01 F1200`

From your file: 
`G1 X-5.605 Y3.69
G1 X-5.786 Y3.454
G1 X-5.975 Y3.225
G1 X-6.171 Y3.002`

The S value tells the viewer how dark to draw the line

openhardwarecoza, 2016-10-23:
(: it will still send it... just wont be able to display it

closed 2016-10-23 by openhardwarecoza

maudette01432, 2016-10-23:
We can keep this closed...I may open another one.  I changed my post to output the S values for every move....still does not load - still does not render...will not send.  The file doesn't show up anywhere in the UI and there is no indication that there is anything to send.

openhardwarecoza, 2016-10-23:
Here's a file you can test with:

[file (1).zip](https://github.com/openhardwarecoza/LaserWeb3/files/546397/file.1.zip)

1.  Open file from the Open Button
2. Got to CAM tab, click Generate GCODE (Since we see any opened file as 'just another layer' - the generate gcode button takes your start/stop params, adds it as start and end gcode, adds all the layers in between, and finally renders it.  it also loads it up into the queue ready to play

[image]

Tested it just now, and it still works

openhardwarecoza, 2016-10-23:
(: No visual indication: Open the GCODE tab before and after opening the file...
Generate gcode renders the preview, but the file is already loaded onto the GCODE tab after opening

openhardwarecoza, 2016-10-23:
Mind you, even your original file does load after clicking Generate GCODE

[image]

... but... I'm looking for .gcode, not .nc files... I'll add  .nc handling in a minute

openhardwarecoza, 2016-10-23:
OK, added .nc as handled extension.

I also automated the clicking of the Generate GCODE button in the name of userfriendlyness

The catch is going to be if you have a couple large rasters loaded, and then open a gcode file, it will then process all the rasters and then the resultant gcode, may thus take a few mins

openhardwarecoza, 2016-10-23:
Scratch that - its cool now

Git pull, test and let me know

openhardwarecoza, 2016-10-23:
(oops accidentally added some tree-cam work onto that first commit :(

openhardwarecoza, 2016-10-25:
Upload the new file?

maudette01432, 2016-10-25:
Peter.  Per gcode spec  The s value once set doesn't change until it is
overridden.  The viewer should render the line with the last s value used
until the s value changes.

openhardwarecoza, 2016-10-25:
Pull requests to fix that would be welcome:

Look at around: If a line does not have an S - we currently default it to 0.3 (30%)
https://github.com/openhardwarecoza/LaserWeb3/blob/master/public/lib/gcode-viewer/gcode-parser.js#L299

There is some leftovers from the Joe Nwales code here https://github.com/openhardwarecoza/LaserWeb3/blob/master/public/lib/gcode-viewer/gcode-parser.js#L944 where f is defined as lastline's F if its not specified.  You'll need to dig in and do the same, store S as lastline.S and if its not defined, used lastline.S

openhardwarecoza, 2016-10-25:
Mind you there is some lastline.s already.  If its populated, I dont know 
https://github.com/openhardwarecoza/LaserWeb3/blob/master/public/lib/gcode-viewer/gcode-parser.js#L967

openhardwarecoza, 2016-10-25:
(: pull request accepted...

commit 8275789c (2016-11-18): Fixed #163 even more. Only renders loaded gcode, as autoclickingenerate gcode would have made a bad user experience if any large rasters was also loaded