LaserWeb/deprecated-LaserWeb3 #164: Trace from bitmap does not keep the trace

maudette01432, 2016-10-19:
When importing a JPG/BPM and tracing the trace is offset from the bitmap image.  This results in an outline that is very difficult to align with the original image.  This would be highly useful for a "trace+raster" feature that other tools have...in lieu of the current lack of support for a raster of closed vectors.

openhardwarecoza, 2016-10-20:
This isnt the intended purpose of the trace feature.  If you want it this way,  crop out whitespace from your bitnap before tracing.  We cannot programmatically handle whitespace as an 'offset'

closed 2016-10-20 by openhardwarecoza

maudette01432, 2016-10-20:
Peter,

I know you closed this so perhaps we can take this offline.  The tracer isn't likely what's causing the problem it's likely the vector importer.  Is there a way for the vector importer to not drop the offsets present in the source files?  LaserWeb3 appears to pull all the vectors to the lower left (with some odd ball offset) regardless of the offsets present in the SVG/DXF file.  This makes it difficult to use the "Easy" tool flow for generating multiple engraves/cuts from multiple SVG files.  One might be imported with the "CUT" layer and one with the engrave...but then we have to mess with offsets even if they are present in the SVG.  Is there any way to dump the traced SVG that gets generated?

Perhaps we can take this discussion offline and recluse this...I'd just like to see this be as useful as possible to as many people as possible and having an raster+trace feature would be awesome.  If we offset both the raster and the traced SVG by the same amount everything would be good.  There just doesn't appear to be a way to calculate with 100% accuracy what offsets should be put back in to get the traced vector re-aligned with the raster.  Where do the seemingly random offset numbers for the SVG come from?

[image]

maudette01432, 2016-10-20:
An additional comment...I think it's this:
https://github.com/openhardwarecoza/LaserWeb3/wiki/Settings:--Image-Position

Where is this in the code so I can hack at it and see if it works?

Is there a way to turn this off so it just imports with whatever positions are in the DXG/SVG?