LaserWeb/deprecated-LaserWeb3 #171: Enhancement: Upload to USB via nodejs

openhardwarecoza, 2016-10-28:
Stumbled across this cool module https://github.com/resin-io-modules/drivelisthttps://github.com/resin-io-modules/drivelist

- I think we can use this, along with its system property, to detect the Smoothie Mass Storage when run over USB from server-smoothie.js, and then use nodefs to write the gcode files over USB. That will be much faster than the serial upload.  In cases where USB is used (like 99% of LW users) this would theoretically allow us to upload over USB at an acceptable speed

It lists the connected drives in your computer. The resulting drives contain a system property that determines if the drive is removable or not, for example, on Windows:

``` javascript
    {
        device: '\\\\.\\PHYSICALDRIVE0',
        description: 'WDC WD10JPVX-75JC3T0',
        size: '1000 GB'
        mountpoint: 'C:',
        system: true
    },
    {
        device: '\\\\.\\PHYSICALDRIVE1',
        description: 'Generic STORAGE DEVICE USB Device',
        size: '15 GB'
        mountpoint: 'D:',
        system: false
    }
]
```
You can filter the list by the one you're interested in, and inspect the system property:

``` javascript
var drivelist = require('drivelist');

drivelist.list(function(error, disks) {
  if (error) throw error;

  var drive = drives.find(function(drive) {
    return drive.mountpoint === 'C:';
  });

  if (drive.system) {
    console.log('C: is not removable!');
  }
});
```They use various mechanisms to determine if a drive is removable, feel free to take a closer look on the platform you that you're most interested in!

DarklyLabs, 2016-10-29:
Awesome!
This will definitely provide users with much more reliable and consistent results than streaming over USB or WIFI.

openhardwarecoza, 2016-11-21:
Alright, I did some experimentation with this on github.com/openhardwarecoza/CoPrint

USB Drive view
[image]

Upload to SD
[image]

Evn "View Config"
[image]

Busy adding Play + firmware update too

openhardwarecoza, 2016-11-27:
So, sadly, after +Wolfmanjm's comments in thread here https://plus.google.com/+PetervanderWalt/posts/BMkq52ME4DW?sfc=true I'll close this one as WONT IMPLEMENT

closed 2016-11-27 by openhardwarecoza