LaserWeb/deprecated-LaserWeb3 #174: Life feed and spindle/laser overrides

cprezzi, 2016-11-02:
I got feed and spindle/laser overrides working for Grbl (>1.1d) and Smoothie via USB.
The GUI elements are only shown when a compatible board is connected via USB.

Somebody like to test?

cojarbi, 2016-11-02:
I have a grbl cnc on 1.0, I can update to 1.1

cprezzi, 2016-11-02:
You are welcome to test this version is on branch "overrides".

cprezzi, 2016-11-11:
Overrides branch got an update. 
Long press on +/- override buttons now continously count values up/down and CTRL+button makes larger steps (10% instead of 1%).
Override fields are only shown if a compatible board is connected though USB (GRBL>v1.1 or Smoothieboard).

beikeland, 2016-11-15:
Seems to be working well; but I can't help to think that the % override is less interesting than the absolute value.

I would at least find it more useful to see that I'm running at F: 120 than knowing the G-code is for F100 and I'm at 120%.

So the question becomes how to display the absolute feeds and speeds and maintain the override control without cluttering the interface, maybe something like so which also adds individual zeroing for axis?
[image]

openhardwarecoza, 2016-11-27:
@beikeland since F and S actual valus changes constantly during the job (especially in rasters) the override percentage makes more sense.

openhardwarecoza, 2016-11-27:
merged to master... closing this issue

[image]

closed 2016-11-27 by openhardwarecoza