LaserWeb/deprecated-LaserWeb3 #176: Smoothie M999 reset after error/end stop hit

Pooky1306, 2016-11-04:
It seems that after smoothie goes to halt mode, LW3 can't send commands anymore.
However the M999 command is required to clear the halt state but it just goes in the buffer. It would be cool to be able to send the M999 command  from LW.

Btw hard reset is working but in some case a restart of node.js is required.

openhardwarecoza, 2016-11-05:
Agreed! Will do asap

openhardwarecoza, 2016-11-27:
Implemented today

Please test!

[image]

closed 2016-11-29 by openhardwarecoza

cprezzi, 2016-11-29:
Seems to work perfect :)
Hint: you should also dump the grblBufferSize array when you dump gcodeQueue, otherwise the character counting would get messed up. I have corrected that with my last commit.