LaserWeb/deprecated-LaserWeb3 #178: Started an optimisation repo

lautr3k, 2016-11-10:
I wanted to integrate lw.rasterizer but I can not for now... And to be honest I am also lost on the v4 ... So I begin to modularize and optimize the V3. My attempt is to scope all "module", starting with simple things like menu and localStorage.

https://github.com/LaserWeb/LaserWeb3/tree/optimization

jorgerobles, 2016-11-10:
I'm currently testing your rasterizer and is amazing. Superb job!

lautr3k, 2016-11-10:
Thanks @jorgerobles

cprezzi, 2016-11-10:
@lautr3k : You are right, there is a lot of optimization potential in LW3. I was also thinking about some optimizations, especially better separation of frontend and backend. I'd like to push all communication with controller boards to the backend, so the frontend only communicates with the backend and nothing else.

I hope we could combine our work later on.

openhardwarecoza, 2016-11-10:
I think its a good idea.  Stepping stone between where we are and we are
headed.  Would make the final porting of modules to react easier.  One
change at a time @tbfleming

tbfleming, 2016-11-10:
* ```reducers/operation.js```: add missing fields to ```operationBase``` 
* ```components/operation.js```:
  * add new operation type or edit existing type in ```const types```
  * add missing fields to ```const fields```
  * test UI to make sure it handles new operation correctly
* ```lib/cam-gcode.js```: ```getGcode```:
  * add an ```else if``` to the ```'Mill '``` conditional that calls a new function
  * create a new function which handles your new operation
  * pass ```docsWithImages``` to your function. It's an array of object.
  * See ```loadImage``` in ```reducers/document.js``` to see the fields in these objects.
    There is one field not listed since it comes from elsewhere: ```image```. This is a browser Image object.

Testing note: turn on the ```Perspective``` checkbox below the Workspace to be able to drag documents 
around the screen. I need to fix that for the orthoganal camera.

lautr3k, 2016-11-10:
@tbfleming maybe the wrong topic ?

openhardwarecoza, 2016-11-10:
I think that was notes to you on whats needed to integrate the
lw.rasterizer into lw4: ie add raster engrave as an operation

openhardwarecoza, 2016-11-10:
@nraynaud  - https://github.com/LaserWeb/LaserWeb3/issues/178#issuecomment-259724085 would thus be instructions if you would like to help out porting your Dragknife code (and other useful operations) into https://github.com/LaserWeb/LaserWeb4

lautr3k, 2016-11-10:
Splited index template in smaller parts #17076c6
See how the index file look right now https://github.com/LaserWeb/LaserWeb3/blob/optimization/public/index.html#L31

nraynaud, 2016-11-10:
@openhardwarecoza I am on a contract at the moment, but I am still thinking about helping.

openhardwarecoza, 2016-11-10:
All cool! Just posting notes as I think about you - might be helpful later when you need to refer back (:

tbfleming, 2016-11-10:
> I think that was notes to you on whats needed to integrate the lw.rasterizer into lw4: ie add raster engrave as an operation

Yes. I assumed that's why you pinged me.

tbfleming, 2016-11-10:
It looks like we may have a clash with the term 'engrave'. In jscut, it means follow an outline without offset (engrave vs. inside vs. outside). If we use 'engrave' to mean 'raster', then we need a different term for following outlines.

openhardwarecoza, 2016-11-10:
So far its always been called "raster" in lw1-3

cprezzi, 2016-11-10:
follow a line could call on-line ;)

funinthefalls, 2016-11-14:
Almost all the software I have seen uses the term engrave to describe a raster operation, cut for vector operations like following an outline. Would naming the three different vector operations "Outside", "Center", and "Inside" and naming the engrave operation "Raster" be acceptable to everyone as a naming convention? If not, what are some other suggestions?

openhardwarecoza, 2016-11-30:
Although this thread is lacking some updates the descriptions in https://github.com/LaserWeb/LaserWeb3/commits/optimization is amazing. Go check it out and ask @lautr3k where he needs help

openhardwarecoza, 2016-12-13:
@lautr3k - just checking in for a status update
As the cam portion of LW4 made big strides since the start of this, i need to start thinking at which point we cross over from working extensively on lw3 to focussing on lw4
All lw4 is missing is comms and raster really

openhardwarecoza, 2017-03-05:
Saddest issue to close - but looks like all that work for nothing... As we have LW4 ready to go.  Thanks again @lautr3k - that definately helped along a lot of the other steps!

openhardwarecoza, 2017-04-13:
Reopening the issue (:

An opportunity arose to reuse this branch! Yay! All that code WAS NOT FOR NOTHING after all!

closed 2018-01-29 by jorgerobles