LaserWeb/deprecated-LaserWeb3 #180: "Laser test" button for jog-tab

cprezzi, 2016-11-11:
It would be usefull to have a "weak laser" button to manually fire the laser with low power. This is often used on diode laser engravers for positioning. The power of such a test shot should be configurable in the settings (weak laser power).

funinthefalls, 2016-11-14:
Wolfmanjm added this to the cnc-firmware bin, however I have an issue with the way he implemented it. You can pulse a low power test fire, but it stays on until you issue an off command for it, which I find to be very awkward to do, especially when trying to align mirrors. A work around is to make a macro that fires on a short Z axis move, but that causes conflicts with those that have adjustable Z axis platforms, or a Z axis slide. I would love to see your idea for a weak test fire button of a very short duration be added.

cojarbi, 2016-11-14:
Anthony remember he did that based on his diode Laser setup.

iceman1979, 2016-11-14:
Can this be changed to a click on / click off behavior?

funinthefalls, 2016-11-14:
When aligning mirrors on a CO2 laser, the pulse duration should only be about 1/10th of a second. Impossible to click n and off that fast.

openhardwarecoza, 2016-11-14:
Easy enough to do with js: fire 10; dwell 0.1; fire 0" or whatever the
syntax.  Does Smoothie have 'dwell?'  @wolfmanjm

wolfmanjm, 2016-11-14:
dwell would be G4 P100  (milliseconds) or G4 S2 (seconds).

also it would be better to be fire 10 then  G4 P100  then fire 0ff

iceman1979, 2016-11-14:
Well, that explains why that pulse feature exists in another software program I use :) The author of that program also has a CO2 laser :) I don't know what syntax he uses to do this but I have tested it on my O'scope.

Sounds like something we might be able to do in software though.

cprezzi, 2016-11-16:
When we adjust the mirrors on the Lasersaur we just cut a short line with low power, as we don't have a test button there.
I like the solution that the host software can activate and deactivate the laser (without move). This way, the duration can be configured in the software, depending on the needs.

cprezzi, 2016-11-16:
I see two different usages:
- On CO2 lasers, it would be useful to have a test shot button, which fires for preconfigured time and power.
- On diode lasers, it would be better to have a toggle button which switches the laser on or off at preconfigured power.

We could add two settings values like "test fire power" and "test fire duration" and make duration 0 to be permanent on/off to hande both usages.

But this feature depends on the firmware implementations.

cprezzi, 2016-11-21:
I have coded this feature in branch "overrides". Please test :)

cprezzi, 2016-11-21:
You must set "Laser Test Power" and "Laser Test Duration" in G-Code Settiings first.

- A duration of zero means the <Laser Test> button acts as a toggle on/off button. 
- If duration is set to more than 0, one button click fires the laser for this duration in ms.

cprezzi, 2016-11-21:
@chamnit: Does Grbl also support G4 for delay (like Smoothie)? At the moment I have coded it with a setInterval on the host side, but would prefer to let grbl handle the timing.

cprezzi, 2016-11-21:
Ok, found it. G4 Px on Grbl uses seconds (not milliseconds, like Smoothie). Changed it.

openhardwarecoza, 2016-11-27:
Merged to Master. Closing Issue

closed 2016-11-27 by openhardwarecoza