LaserWeb/deprecated-LaserWeb3 #185: motors off button

quillford, 2016-11-19:
personally, I use M18/M84 a lot. would be convenient to have it above the jog buttons. I can add it myself, but would other people find this useful? maybe it can be a single macro button that exists on the jogging panel so it can be configured for laser fire or whatever

maudette01432, 2016-11-19:
I thing taking a page from usgs would make a lot of people happy.  There is a pane of several customizable macro buttons which will send what ever command(s) the user has programmed for the macro.

quillford, 2016-11-19:
I know about the macro panel, but it would be convenient to have macro button 0 for example appear in the jogging pane. I frequently use motors off to move the carriage out of the way or for rough positioning.

cojarbi, 2016-11-19:
We already have a macro panel. Why dont you place that motor off on the end
gcode?

openhardwarecoza, 2016-11-20:
Well,  the limitation a while back was 'please don't add anything more to
the jog panel,  since @FabCreator wants to run off a 1024x800 7" Pi
Monitor,  and we dont want to have a scrollbar for their users.

Since then,  i talked them into not using Pi+7"tft. They are now running on
a Pipo X9S -  lookup the specs of the screen and just test that we still
fit nice and finger touchable sizes in their resolution (dev tools -  view
port)  and you are welcome to add some more!

openhardwarecoza, 2016-11-20:
> nt-261740692>,

Oh just one more thing, but please do a check if we are connected to
Smoothie before showing such a hard coded button, since Grbl doesnt have a
Motors off command (only motor off after delay)

quillford, 2016-11-20:
then maybe it should be configurable so macro 0 from the macro panel appears on the jog panel instead of a fixed motors off, especially since it seems motors off isn't too widely used

FabCreator, 2016-11-20:
For smoothie users moving the motors is not a good idea too fast and you will damage the drivers. I would recommend avoiding it and jogging.

Perhaps the jogging options could pop up like the number pad does as an option?

openhardwarecoza, 2016-11-27:
@quillford we now have firmware specific code here https://github.com/LaserWeb/LaserWeb3/blob/master/public/js/socketcomms.js#L24-L31 that you are welcome to plug into to show/hide the Motors off button

openhardwarecoza, 2016-11-27:
[image]

Done!

commit f8cc59c1 (2016-11-27): Jog UI - added Reset buttons for F,S and Zero Axes, and also added Motors Off button for fix #185
closed 2016-11-27 by openhardwarecoza