LaserWeb/deprecated-LaserWeb3 #188: Streaming Protocol for GRBL

cprezzi, 2016-11-25:
I have implemented the grbl streaming protocol in branch overrides. Could somebody please test.

Should I make the protocol swichable between ping-pong and streaming, or is it ok to move to the streaming protocol completely?

openhardwarecoza, 2016-11-25:
@chamnit would probably say Go Streaming! (:
This is awesome!

openhardwarecoza, 2016-11-27:
I've pulled over overrides into Master so we can get a couple more users testing

I accidentally grabbed some TinyG work (nice coincidence, my new controller from Synthetos should arrive this week... the new G2 based laser specific controller of theirs...)

I cleaned up a error in server-smoothie.js too

closed 2016-11-27 by openhardwarecoza

cprezzi, 2016-11-27:
Good you checked server-smoothie as I was not able to test because the lack of a smoothie board.

cprezzi, 2016-11-27:
I started working on TinyG but didn't came to far. I try to use the comm library from TinyG but will need a board to finish work.