LaserWeb/deprecated-LaserWeb3 #191: Feed and Power overrides - max 200% ?

funinthefalls, 2016-11-29:
I have noticed that the real time overrides are limited to 200%, is there any specific reason why this has been limited to this value?

cprezzi, 2016-11-29:
Override rages are limited by the firmwares:

- Grbl limits the feed override range from 1% to 200% and the spindle/laser override to 50-200% (some of these values are changeable in grbl's config.h).
- I did not find the valid ranges for smoothieware. 

@arthurwolf : What are the valid ranges of M220 and M221 for smoothie?

closed 2016-11-30 by openhardwarecoza

chamnit, 2016-12-05:
Grbl override ranges are somewhat arbitrary. They are based on expected ranges of use. In extreme circumstances, each of them can be altered in the config.h file with a value between 1-255.