LaserWeb/deprecated-LaserWeb3 #195: Emblaser-electron issues: Grbl support

openhardwarecoza, 2016-11-30:
I think this one is better handled with @cprezzi 's plan to have a single server.js for all firmwares. But opening issue so we dont forget

cprezzi, 2016-12-01:
The single server.js is finished. Please test on Smoothie. Testing on Grbl is looking good.

@openhardwarecoza : Probably you want to merge that to your Electron-LW3 branch?

openhardwarecoza, 2016-12-01:
Merged into Electron-LW3 already https://github.com/LaserWeb/LaserWeb3/commits/Electron-LW3
Busy compiling new installer...

openhardwarecoza, 2016-12-01:
Tested on Smoothie, and no issues

openhardwarecoza, 2016-12-01:
New installer ready to test with Grbl... Works with Smoothie https://drive.google.com/file/d/0B-UxBll2MapjX1NWMDVzWWhUbms/view?usp=sharing

closed 2016-12-09 by openhardwarecoza