LaserWeb/deprecated-LaserWeb3 #201: Minor error on connecting to Smoothieboard

funinthefalls, 2016-12-10:
When I connect to my Smoothieboard with LW3, I get the following in the console:

 Version currently Installed : 30476
: Version available online on Github : 30476
: Your version of LaserWeb is Up To Date!
: Build version: edge-dcf0252, Build date: Nov 27 2016 14:18:23, MCU: LPC1768, System Clock: 100MHz
:  CNC Build
: error:Invalid statement

It does not seem to cause any issues, just wondering why it gives the invalid statement error?

cprezzi, 2016-12-10:
Yes, I know. On connection we send several commands to find out which firmware it is. What you see is Smoothie complaining about a command for TinyG.
I will try to hide such error if possible...

funinthefalls, 2016-12-10:
OK, that makes sense. Thank you.

funinthefalls, 2016-12-28:
I will close this off now, since it really does not impact things at all.

closed 2016-12-28 by funinthefalls

cprezzi, 2016-12-30:
The "false" error message should be gone with the new push from today.

funinthefalls, 2016-12-31:
Thanks @cprezzi . Will test and report back in a few minutes.

funinthefalls, 2016-12-31:
WooHoo! minor error is gone. Thanks for fixing this @cprezzi