LaserWeb/deprecated-LaserWeb3 #205: STL Slice issues

openhardwarecoza, 2016-12-15:
1)  Shapes doesnt seem to always close properly

[image]

2)  Generating GCode looses the offsets from the plating

[image]

openhardwarecoza, 2016-12-15:
@lautr3k - the closing shapes thing might also be what we had with svg/dxf https://github.com/mrdoob/three.js/issues/9450 - I didnt touch your SLAcer.js code in that sense, its still using ShapeGeometry

lautr3k, 2016-12-15:
@openhardwarecoza SLAcer need a "clean" mesh , try to fix it with some tools (MeshLab/Netfabb)

openhardwarecoza, 2016-12-15:
Not just that.  Read the linked thread,  linegeometry works much better in
our application because all we want is the line data

openhardwarecoza, 2017-03-05:
Closing - wont be fixing LW3 anymore

closed 2017-03-05 by openhardwarecoza