LaserWeb/deprecated-LaserWeb3 #212: High CPU load by web browser

jekhor, 2017-01-04:
At LW3 opening, Firefox (and chrome) starts to consume CPU too much and don't stop this until tab will be closed or other tab is activated.

At my system, Debian unstable, Firefox 50.1.0, i7-5600U, Firefox loads CPU for 60%.

openhardwarecoza, 2017-01-04:
We only support Chrome

Try Laserweb.github.io/LaserWeb4 which has a more lightweight webgl implementation.  Lw3 uses three.js which tends yo need good hardware and good webgl performance (Linux gfx drivers and webgl has always lagged a bit too)

openhardwarecoza, 2017-01-04:
Besides,  as we move into LW4 all the more,  its unlikely we'll massively improve LW3.  It works for most people

closed 2017-01-04 by openhardwarecoza

jekhor, 2017-01-04:
The same on LW4, btw.