LaserWeb/deprecated-LaserWeb3 #213: Cuttings order optimization

Biondilbiondo, 2017-01-04:
Using laserweb with our self-made lasercutter we have noticed that frequently the gcode generated by the "simple cam" of laserweb have a lot of usless G0 move. In our statistics we found out that in typical gcodes only between 10 and 50 % of the distance covered by the machine is for cutting. I have implemented a simple algorithm that allow to get a much more efficient gcode, drastically cutting the usless movement of the machine around the cutting area. I hope that this could help for get more fast and accurate cuttings but probably a better implementation is needed for a full support in laserweb. If you want to see the code you can find it here: https://github.com/Biondilbiondo/LaserWeb3
Thanks

openhardwarecoza, 2017-02-15:
Well, as we are about to launch LW4 (very very soon) - I am closing this one since that work will be redundant anyway

closed 2017-02-15 by openhardwarecoza

openhardwarecoza, 2017-03-24:
Any chance you can bring your protocol onto LW4?