LaserWeb/deprecated-LaserWeb3 #214: Log-to-file request (not a bug)

donkjr, 2017-01-04:
I found the console output [c:\program files\nodejs\node.exe] while LW is running very useful for troubleshooting, but the screen scrolls until history is out of view.
An option to push this to a log file would be a great troubleshooting feature.

tbfleming, 2017-01-05:
On Linux you can ```node args 2>&1 | tee log_filename```

openhardwarecoza, 2017-01-05:
On windows http://www.robvanderwoude.com/battech_redirection.php

closed 2017-01-05 by openhardwarecoza

cprezzi, 2017-01-06:
I will probably implement this in the next version.

cprezzi, 2017-01-06:
I have implemented server logfile in the new version (can be switched on in config.js).  
Not perfect yet, but a good start.

donkjr, 2017-01-07:
Nice and thanks for your consideration of this idea, it will help a a lot when troubleshooting .... 

BTW .....
After a lot of hacking I got a batch to write to a file but you loose the console output.
LW3start_Log.bat
   cd c:\laserweb3 
start "LW Started with log file [LW3log] on Desktop" cmd /c "node server.js ^1>        c:\Users\Owner\Desktop\LW3log.txt"
   start chrome http://localhost:8000/

Seems that when running from the START cmd the redirection is associated with the START command itself not the following command. The escape method "^" did not work for me...