LaserWeb/deprecated-LaserWeb3 #226: Serial Port

btcrowe, 2017-01-28:
I have installed Laserweb3 and it opens, but i can not connect to the laser. It won't let me select com port 4 which is the one my laser is on. It doesn't list any ports in the select port pull down.

openhardwarecoza, 2017-01-28:
Git is for Bugs. Not for support.  Head over to the google plus community 

If you failed to read the first line of the Readme (telling where to go to get support)  you probably also didnt read the install instructions.  Or you failed to read the syslog during startup (errors about no webgl?  Then that pc doesn't support LW...)

I can point you to reading lessons if that might help

openhardwarecoza, 2017-01-28:
Scroll to the top of the syslog window in lw

openhardwarecoza, 2017-01-28:
Scroll to the top of the syslog window in lw's web interface

closed 2017-01-28 by openhardwarecoza