LaserWeb/deprecated-LaserWeb3 #229: 404 while getting serialport-v4.0.7-node-v48-linux-arm.tar.gz

radimklaska, 2017-01-31:
Hi, I'm trying to install LaserWeb3 on Raspberry Pi by following instructions here: https://github.com/LaserWeb/LaserWeb3/wiki/Install:-Raspberry-Pi-Image-Build

```
pi@rk-pi2:~ $ npm -v
3.10.10
pi@rk-pi2:~ $ node -v
v6.9.4
pi@rk-pi2:~ $ cd LaserWeb3/
pi@rk-pi2:~/LaserWeb3 $ npm install

> serialport@4.0.7 install /home/pi/LaserWeb3/node_modules/serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v48-linux-arm.tar.gz
```

Not sure how to get around that.

cprezzi, 2017-01-31:
It seems the download link for the serialport lib is not working (at the moment). You can try `npm install serialport` or try later.

cprezzi, 2017-01-31:
I close this issue, as it is not LaserWeb specific.

closed 2017-01-31 by cprezzi