LaserWeb/deprecated-LaserWeb3 #231: laserweb3 no port

gbrinki, 2017-02-15:
after installation and settings i have no possibility to choose a serial port.
second probllem
[image]
wich pin is asked to get the mashine started?

tbfleming, 2017-02-15:
@openhardwarecoza This is a recurring user error. Maybe the PIN should default to disabled. I doubt the PIN is good for safety since machines tend to have a physical test laser button.

openhardwarecoza, 2017-02-15:
This particular instance is an JS error somewhere in the stack (if pin is a text field instead of how it looks if the jquery plugin ran)

When theres JS errors in the stack, things stop working (ie hiding pin, threejs, jquery for the port selection, etc) 

So @gbrinki , read the syslog near the top and tell us what it says

openhardwarecoza, 2017-02-15:
If there were no errors (webgl started, etc) the Pin field renders like this when the JQuery plugin takes it over... 

[image]

openhardwarecoza, 2017-02-15:
Ctrl+shift+i console during a refresh may tell you more too
[image]

If your log has errors, it will do what i see above

openhardwarecoza, 2017-02-15:
Now, closing this as its not a bug, just a plain support question, and the README does tell you where to go for that

[image]

(If you didnt read the readme, what else didnt you read (;   - maybe thats why its not working.  Ie must be Chrome, must be on a machine with good webgl support, etc)

closed 2017-02-15 by openhardwarecoza

gbrinki, 2017-02-17:
thank you for your mail. Now I have made 3 screenshots of my Laserweb3
console. I do not know exactely but i think there have been found two
mistakes but i do not know how to handle them and would be glad if you can
help me, Thanks!

gbrinki, 2017-02-17:
[image]
[image]
[image]

tbfleming, 2017-02-17:
@gbrinki Please reread the text @openhardwarecoza highlighted in yellow before he bans you.

openhardwarecoza, 2017-02-17:
(; yeah I am getting close... 
Itching on the ban button

Anyway, thats a browser plugin preventing websocket access... Use a clean install