LaserWeb/deprecated-LaserWeb3 #232: Laser Problem

Fred9543, 2017-02-17:
Hello, 

I don't know if this is the right place but, I will try to ask anyway

I have been trying to make this "Laser Engraver" work with Laserweb3 http://www.ebay.com/itm/5-5W-30X40cm-Laser-Engraving-Machine-Engraver-Wood-Cutter-A3-DIY-Logo-Printer-/161937138736?hash=item25b434a430:g:vo0AAOSw5dNWijye 

I have now been trying for like 2 days with no solution. I have used a lot of different programs, benbox and eleksmaker, those won't get my laser to work. And I got the same problem when using LaserWeb, but when using Laserweb the movement of the axis works really well. The only problem is that my 5500mW laser lights up always with low light, and when i start the engraver the laser dosn't burn anything (stays at lowlight). It says "Laser on/off missing" but i can't find any where to type in laser settings in the Laserweb3 software (I have looked in the settings but it seems to be totally gone). What am I doing wrong?. Sorry if it is a stupid question i'm new at this and just want to understand how I can get this Grbl machine to work. My main borad is a Eleksmaker Mana SE, which should be PWM already from the production. 

Thanks in advance.

openhardwarecoza, 2017-02-17:
[image]

openhardwarecoza, 2017-02-17:
Git is for bugs. Plus is for Questions

Furthermore though, thats a hardware question, not a laserweb question:  Check, your diode probably has a shit cheap driver that doesnt support PWM, or you forgot to depress the "Low light" switch on the controller

closed 2017-02-17 by openhardwarecoza