LaserWeb/deprecated-LaserWeb3 #234: [Question] Drag Knife usage

lollotek, 2017-02-22:
Hi, I'm using LaserWeb3 to generate the gcode for my cnc.
I'm trying to use a drag knife, and I see the option into LaserWeb. 
The strange thing is that I don't see any changes on resulting gcode, but I expecting the arcs into edges for blade alignment.
Could be that I need other settings? I already enabled CNC mode. Seems to me, also others options like outside/inside doesn't changes the resulting file.
Thanks

lollotek, 2017-02-22:
Now I see the error into console after pressing 'Add' Button:

> Uncaught TypeError: Cannot read property 'color' of undefined
>     at dragknifePath (threegcode.js:583)
>     at addOperation (threegcode.js:336)
>     at HTMLButtonElement.onclick (VM1318 :1)

threecode.js:583 (master branch)
`dragknifeGrp.userData.color = dragknifeGrp.material.color.getHex();`
because dragknifeGrp.material is undefined, I fast solved using costant hex value, and it woked.

openhardwarecoza, 2017-03-05:
Dropped support for it in LW4 anyway. Sorry

closed 2017-03-05 by openhardwarecoza