LaserWeb/deprecated-LaserWeb3 #235: simple 2d gcode visualize

mkeyno, 2017-03-11:
hi , sorry to open ticket for my question , actually I made my low weight web interface on ESP8266 which is use the JavaScript readerFile to load the gcode file on client side , I use the GRBL as cnc interpreter and connected to my ESP and its work well, however I intend to add visualize  tab page so I could show gcode tool path but I couldn't find single file JS code , I really appreciated if I have any hint on how to use essential laseweb3 code on this purpose
[image]

funinthefalls, 2017-03-11:
I like your interface.

openhardwarecoza, 2017-03-11:
Lib/gcode-viewer

closed 2017-03-11 by openhardwarecoza

mkeyno, 2017-03-11:
thanks @funinthefalls , actually I just connect serial pin of Arduino board(GBRL firmware) to ESP module and let the client side page and then ESP manage sending the Gcode to arduino and handle responds. My idea is just connect to CNC and load  file and do CNC job. With `FileReader()` method I could load and buffering file with size of 50K lines with no problem. Web host can authenticate the user and handle only one IP to control the CNC. My page code size less than 20KB and I don't want to use jquery due its size, that's why I'm looking for JS code to embed gcode visualizer in webpage. To enhance the ESP performance I should merge the file together and keep the size less as possible