LaserWeb/deprecated-LaserWeb3 #236: LaserWeb3 failing to install on RPI 2

nikki-reprap, 2017-03-20:
LaserWeb3 is failing to install on RPI 2 I have followed the instructions in the wiki but I get this error

root@raspberrypi:/home/pi/LaserWeb3# npm install

> serialport@4.0.7 install /home/pi/LaserWeb3/node_modules/serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyA                                                                             dvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v48-linux                                                                             -arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@6.1                                                                             0.0 (node-v48 ABI) (falling back to source compile with node-gyp)
gyp WARN EACCES user "root" does not have permission to access the dev dir "/roo                                                                             t/.node-gyp/6.10.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/LaserW                                                                             eb3/node_modules/serialport/.node-gyp"
make: Entering directory '/home/pi/LaserWeb3/node_modules/serialport/build'
make: *** No rule to make target '../.node-gyp/6.10.0/include/node/common.gypi',                                                                              needed by 'Makefile'.  Stop.
make: Leaving directory '/home/pi/LaserWeb3/node_modules/serialport/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_module                                                                             s/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_proces                                                                             s.js:215:12)
gyp ERR! System Linux 4.4.50-v7+
gyp ERR! command "/usr/bin/nodejs" "/usr/lib/node_modules/npm/node_modules/node-                                                                             gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/home/pi/LaserWeb3/                                                                             node_modules/serialport/build/Release/serialport.node" "--module_name=serialport                                                                             " "--module_path=/home/pi/LaserWeb3/node_modules/serialport/build/Release"
gyp ERR! cwd /home/pi/LaserWeb3/node_modules/serialport
gyp ERR! node -v v6.10.0
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute '/usr/bin/nodejs /usr/lib/node_                                                                             modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --mo                                                                             dule=/home/pi/LaserWeb3/node_modules/serialport/build/Release/serialport.node --                                                                             module_name=serialport --module_path=/home/pi/LaserWeb3/node_modules/serialport/                                                                             build/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/pi/LaserWeb3/node                                                                             _modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:877:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/chi                                                                             ld_process.js:226:5)
node-pre-gyp ERR! System Linux 4.4.50-v7+
node-pre-gyp ERR! command "/usr/bin/nodejs" "/home/pi/LaserWeb3/node_modules/ser                                                                             ialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/pi/LaserWeb3/node_modules/serialport
node-pre-gyp ERR! node -v v6.10.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.32
node-pre-gyp ERR! not ok
Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/npm/node_modules/node-g                                                                             yp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/LaserWeb3/node_mo                                                                             dules/serialport/build/Release/serialport.node --module_name=serialport --module                                                                             _path=/home/pi/LaserWeb3/node_modules/serialport/build/Release' (1)
laserweb@3.256.0 /home/pi/LaserWeb3
├─┬ chalk@1.1.3
│ ├── ansi-styles@2.2.1
│ ├── escape-string-regexp@1.0.5
│ ├─┬ has-ansi@2.0.0
│ │ └── ansi-regex@2.1.1
│ ├── strip-ansi@3.0.1
│ └── supports-color@2.0.0
├── dotenv@4.0.0
├─┬ node-static@0.7.9
│ ├── colors@1.1.2
│ ├── mime@1.3.4
│ └─┬ optimist@0.6.1
│   ├── minimist@0.0.10
│   └── wordwrap@0.0.3
├─┬ request@2.81.0
│ ├── aws-sign2@0.6.0
│ ├── aws4@1.6.0
│ ├── caseless@0.12.0
│ ├─┬ combined-stream@1.0.5
│ │ └── delayed-stream@1.0.0
│ ├── extend@3.0.0
│ ├── forever-agent@0.6.1
│ ├─┬ form-data@2.1.2
│ │ └── asynckit@0.4.0
│ ├─┬ har-validator@4.2.1
│ │ ├─┬ ajv@4.11.5
│ │ │ ├── co@4.6.0
│ │ │ └─┬ json-stable-stringify@1.0.1
│ │ │   └── jsonify@0.0.0
[... 86 lines cut ...]
    └── json3@3.3.2

npm ERR! Linux 4.4.50-v7+
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install"
npm ERR! node v6.10.0
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE

npm ERR! serialport@4.0.7 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@4.0.7 install script 'node-pre-gyp install --f                                                                             allback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/LaserWeb3/npm-debug.log

openhardwarecoza, 2017-03-20:
Install serialport@latest manually 'npm install serialport@latest' and continue step after npm install

openhardwarecoza, 2017-03-20:
Closing. Lw3 is slowly being end of life'd for lw4.
Pi isnt yet a supported platform of lw4 (later once lightweight interface/server is ready around 4.2.x)

closed 2017-03-20 by openhardwarecoza