LaserWeb/deprecated-LaserWeb3 #238: Feature Request - Simulation

OzRon, 2017-03-28:
Hello.  We would like to open a gcode file and test it by simulation preview before running it on the cutter.  To be able run the simulation without being connected to the machine is preferred.  A different color for the tool path showing the simulated cut would be good.  Thanks & keep up the great work!

analogmonster, 2017-03-28:
Trust me, this issue will be closed as soon as he gets to it. Just switch to laserweb 4, simulation is there already. The laserweb4 binaries makes it really easy to upgrade and you could always keep 3 on your system as a backup until you're sure you want to keep 4. You will, its awesome!

openhardwarecoza, 2017-03-28:
@analogmonster you are correct sir, and earned my redpecy

closed 2017-03-28 by openhardwarecoza