LaserWeb/deprecated-LaserWeb3 #240: Issue with zig-zag rasterisation of bitmap

andreword, 2017-09-16:
Though this may be down to my configuration, I have found that when rasterising a PNG file on both Laserweb 3 and 4 (1000 dpi image). The 'reverse' path is offset from the normal direction, resulting in a ghosting effect.(see image, and ignore the scratches, it's a piece of scrap copper board)

[image]

I modified the public/js/laserraster.js to remove the line
    this.dir = -this.dir; // Reverse direction for next row - makes us move in a more efficient zig zag down the image

Though less efficient, this results in the program drawing all raster lines from the same side, and thus removing this ghosting effect, resulting a solid image

If this is down to the file geometry or some other setting I haven't found let me know

An option to disable the zig-zag pattern would be useful

I haven't managed to test this on laser web 4 as yet

cprezzi, 2017-09-17:
This is not a software issue, it's your hardware that has backlash. You should try to improve the hardware.

andreword, 2017-09-17:
Well I'm afraid it is a software issue if the same hardware is used to render with other programs and does not suffer the same issue when rendering in the zig-zag pattern.
Neither does this problem appear on vector movements where gaps would appear in lines

Could you tell me what settings you are using, in particular the dpi of the images you import

tbfleming, 2017-09-17:
@andreword this is a support issue, not a bug. Ask at https://plus.google.com/communities/115879488566665599508

Most likely cause: your firmware's acceleration or max rate settings are too high. When this happens some things work and some things don't.

andreword, 2017-09-17:
The image in my post above was taken after running the engraver at slow speed to increase burn time on the painted surface. The engraver also runs at speeds >8 times the speed used for this particular engraving without any issues (except for bitmaps).

When I disable the zig-zag pattern, and only burn in a single direction, the result is as per the attachment
[image]

If the issue was missing steps or high speeds the effect would also be visible in this image, but it isn't

tbfleming, 2017-09-17:
Look at the generated gcode. If it's a SW issue then the issue will show up there. If you can't read gcode, then run it through an emulator.

closed 2018-01-29 by jorgerobles