LaserWeb/deprecated-LaserWeb3 #241: Installation Issue

UkieII, 2017-11-25:
I am relatively new to LaserWeb and am having a problem installing the software on Windows 10. I previously had it installed and was adjusting one of the settings when the program froze up and I had to close it using Task Manager. Once I did that, the program would start and all I can get now is a blank white screen and not much else. I'm not sure whether I somehow created a conflict with some other software on my system or whether it is a virus or whatever. I have also tried deleting and reinstalling the program to no avail. Has anyone else had this problem and if so, can I get some help in resolving my problem. Thanks.

cojarbi, 2017-11-25:
Please read this post a Gplus . You will find a link to where it explains how to fix this https://plus.google.com/+ArielYahni/posts/dr7R34GUidY

closed 2017-11-25 by cojarbi

UkieII, 2017-11-25:
I fixed the problem with your help. Thanks again.

   
Closed #241.—